ArduCAM/MIPI_Camera #199: Imx477 Camera drivr doesnot work properly with Jetson orin nano

SharangShakti, 2024-06-20:
I have a Jetson Orin nano
Release : 5.10.120-tegra
L4T: 35.4.1
Jetpack: 5.1.2

I installed the driver :[arducam-nvidia-l4t-kernel-t234-nano-5.10.120-tegra-35.4.1-20230809074912_arm64_imx477.deb](https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.1-orin-nano/arducam-nvidia-l4t-kernel-t234-nano-5.10.120-tegra-35.4.1-20230809074912_arm64_imx477.deb)

Then I ran :
` v4l2-ctl --list-devices`
which Outputs 
```
NVIDIA Tegra Video Input Device (platform:tegra-camrtc-ca):
	/dev/media0

vi-output, imx477 9-001a (platform:tegra-capture-vi:1):
	/dev/video0
```
But when i try to run :
`gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=1920,height=1080,framerate=30/1' ! nvvidconv ! 'video/x-raw,format=(string)I420' ! xvimagesink`
A window pops up with black screen 
and the log looks like this
```

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 4032 x 3040 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3840 x 2160 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 2 
   Output Stream W = 1920 H = 1080 
   seconds to Run    = 0 
   Frame Rate = 59.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadExecute:694 NvBufSurfaceFromFd Failed.
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:247 (propagating)
WARNING: from element /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0: Pipeline construction is invalid, please add queues.
Additional debug info:
gstbasesink.c(1209): gst_base_sink_query_latency (): /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0:
Not enough buffering available for  the processing deadline of 0:00:00.015000000, add enough queues to buffer  0:00:00.015000000 additional data. Shortening processing latency to 0:00:00.000000000.
Got EOS from element "pipeline0".
Execution ended after 0:00:04.004109467
Setting pipeline to NULL ...
GST_ARGUS: Cleaning up
GST_ARGUS: Done Success
Freeing pipeline ...
```
What am I doing wrong?

AhmadRfaie, 2024-12-04:
Hello, any updates on this issue?

hunter55555, 2024-12-13:
try the command
`gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=1920,height=1080,framerate=30/1' ! queue ! nvvidconv ! 'video/x-raw,format=(string)I420' ! xvimagesink`