ArduCAM/MIPI_Camera #208: ./Install_dependencies_jetson.sh  Error    Cannot find the corresponding deb package, please send the following information to support@arducam.com Kernel version:  4.9.337-tegra-32.7.6-20241104234540 Jetson type:  NVIDIA Jetson Nano Developer Kit

DimIsaev, 2025-01-23:
./Install_dependencies_jetson.sh

Error 

```bash
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.337-tegra-32.7.6-20241104234540
Jetson type:  NVIDIA Jetson Nano Developer Kit
```

env

[image]

gluckez, 2025-01-29:
I have the same issue, seems it only supports up to jetpack version 32.7.5?

DimIsaev, 2025-02-05:
@ArduCAM @iliesaya @Hermann-SW  FIY

DimIsaev, 2025-02-12:
> I second this, is there an expected update for L4T 32.7.6 for the following releases: https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1

@SeeRich  in private correspondence with support they promised to release an update after 12.02 (holidays)

DimIsaev, 2025-03-28:
**expectation**

DimIsaev, 2025-04-22:
expectation